pantadeusz/raspigcd #10: Minimal API documentation

pantadeusz, 2018-06-17:
The API should be documented. Especially:

 * [ ] how to use the library (compile project based on it, and "hello world")
 * [ ] crucial functions and data structures
 * [ ] supported g-code commands with information about its usage

DarkwallM, 2019-07-13:
I am building a raspigcd2 latest on a RPi3B+. I look forward to the steps for compiling the project. What platform are you building on?

I am up to date on the RPi software and been trying to build on the RPi. I have been adding software as required. But now it is a versioning issue for the gcc, cmake and doxygen. I am not sure which SDL2 to pull.

Again I look forward to what platform and the versions of the tools you are using.

pantadeusz, 2019-07-13:
You need:

* Debian Buster (raspbian)[https://www.raspberrypi.org/downloads/raspbian/].
* g++-8 or g++-7
* cmake

Optional (mostly if you build on desktop to test it in simulation mode):

* libsdl2-dev

Optional (if you want to build source code documentation):

* doxygen
* graphviz

Procedure can look like (assuming debian buster):

```bash
sudo apt install g++-8 cmake git
git clone https://github.com/pantadeusz/raspigcd
cd raspigcd
mkdir build
cd build
cmake .. # don't worry about sdl and doxygen warning 
make gcd # there are some warnings - you can ignore them. I will fix that later
```

Now you should have the compiled version. When you also do

```bash
make package
```

than you will also have installation version in deb package.

You can run it directly ( ```sudo ./gcd -c ../v4.json -f yourfile.gcd``` ) or install and then use from CLI anywhere in the system. Remember about sudo. It needs to have direct access to GPIO registers.

Note that the configuration file should be properly modified for your needs. The example configuration for my machine is in v4.json. The most crucial part are the pin numbers. The numbering is the broadcom version of numbers. See (pinout.xyz)[https://pinout.xyz/#] BCM.

Please let me know if it worked for you. I am very curious about feedback and testing other than myself.

DarkwallM, 2019-07-15:
Success! I initially built gcd with the first three bullet points. Then I added the other items. During the second cmake it was declared by doxygen  (version 1.8.13) dot missing components mscgen dia. The second build was error free.

Thank you for your help in clarifying the environment.

I will follow up with running gcd after I complete setting up Buster. Thanks again.

pantadeusz, 2022-10-20:
Ok. Since I am working on another approach to the whole project I will close this issue.