pantadeusz/raspigcd #36: Trying to compile raspigcd

spaxxomatic, 2019-10-29:
Hi 
I'm trying to compile the v0.9.1.final.archive, but I'm getting an error using gcc version 4.9.2 (Raspbian 4.9.2-10+deb8u2)  on a Raspi Zero:
```
In file included from /root/raspigcd/v0.9.1.final.archive/src/dto_CoordTranslateConfig.hpp:25:0,
                 from /root/raspigcd/v0.9.1.final.archive/src/dto_CoordTranslateConfig.cpp:21:
/root/raspigcd/v0.9.1.final.archive/src/dto_Steps.hpp: In constructor ‘tp::motor::Steps::Steps(int, int, int, int)’:
/root/raspigcd/v0.9.1.final.archive/src/dto_Steps.hpp:34:46: error: array must be initialized with a brace-enclosed initializer
             : std::array<int, 4>{ a, b, c, d }
                                              ^
/root/raspigcd/v0.9.1.final.archive/src/dto_Steps.hpp:34:46: error: too many initializers for ‘std::array<int, 4u>’
/root/raspigcd/v0.9.1.final.archive/src/dto_Steps.hpp: In constructor ‘tp::motor::Steps::Steps()’:
/root/raspigcd/v0.9.1.final.archive/src/dto_Steps.hpp:38:46: error: array must be initialized with a brace-enclosed initializer
             : std::array<int, 4>{ 0, 0, 0, 0 }
                                              ^
/root/raspigcd/v0.9.1.final.archive/src/dto_Steps.hpp:38:46: error: too many initializers for ‘std::array<int, 4u>’
```

Do you have any idea what might be wrong? 
Thanks

spaxxomatic, 2019-10-29:
OK, I think I'll need an gcc >6. Installed gcc 9.xx, will retry to compile ...

spaxxomatic, 2019-10-29:
.. I came pretty far using g++9, but now the linker will stop with an error:
[ 95%] Linking CXX executable rpigcd
/usr/bin/ld: CMakeFiles/rpigcd.dir/a_main.cpp.o: undefined reference to symbol '_ZTISt9exception@@GLIBCXX_3.4'
/opt/gcc-9.1.0/lib/libstdc++.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/rpigcd.dir/build.make:95: recipe for target 'rpigcd' failed
make[2]: *** [rpigcd] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/rpigcd.dir/all' failed
make[1]: *** [CMakeFiles/rpigcd.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

pantadeusz, 2019-10-29:
that's interesting... Did you try to:

```bash
cmake ..
make clean
make
```

?

pantadeusz, 2019-10-29:
If the problem persists and you would provide more detailed information, then I will look into it tomorrow. I will need:

* debian/ubuntu version
* g++ version
* cmake version
* is catch2 installed (it should be installed)
* raspberry Pi version (I didn't test it on Pi4)
* errors after clean build (git clone, cmake, make, make test)

spaxxomatic, 2019-10-29:
.. came one step ahead trying to link with g++ instead of gcc, but I'm hitting the next error. Will try to 
build everything with g++ and report back 

`-L/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/tpcommon/lib`  -L/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/catch/lib  -L/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/fakeit/lib  -L/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/nlohmann_json/lib -rdynamic libraspigcd.so -ltpcommon -Wl,-rpath,/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/tpcommon/lib:/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/catch/lib:/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/fakeit/lib:/root/raspigcd/v0.9.1.final.archive/build/.cppdeps/nlohmann_json/lib:/root/raspigcd/v0.9.1.final.archive/build
libraspigcd.so: undefined reference to `tp::img::Img8::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
`

spaxxomatic, 2019-10-29:
The answers to your questions: 
debian/ubuntu version: jeesie
g++ version: 9.1.0
cmake version: 3.6.2
is catch2 installed (it should be installed): yes, but compiled with gcc4.3
raspberry Pi version (I didn't test it on Pi4): Pi zero (single core Arm6)
errors after clean build (git clone, cmake, make, make test): see above

spaxxomatic, 2019-10-29:
OK, tried again from scratch with g++ instead of gcc, got the same error:
`libraspigcd.so: undefined reference to tp::img::Img8::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/rpigcd.dir/build.make:95: recipe for target 'rpigcd' failed
make[2]: *** [rpigcd] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/rpigcd.dir/all' failed
make[1]: *** [CMakeFiles/rpigcd.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
`

spaxxomatic, 2019-10-29:
mhhm, the lib is there:  
`nm -gC  /root/raspigcd/v0.9.1.final.archive/build/.cppdeps/tpcommon/lib/libtpcommon.so  | grep save
0000ad08 T lodepng_save_file(unsigned char const*, unsigned int, char const*)
00005618 T tp::img::Img8::save(std::string const&)
00017fd4 T lodepng::save_file(std::vector<unsigned char, std::allocator<unsigned char> > const&, std::string const&)
`
but the signature seems wrong .. I'll delete the cache and rebuild it with g++9 ...

spaxxomatic, 2019-10-29:
That dependency management with node ... don't like it at all, does not fit a C++ dev env

pantadeusz, 2019-10-29:
ahh.. You trying to compile older version. I don't reccomend it. The new version is in main directory of the repository. I should probably remove this directory (will do it soon) :/

Try the following:

```
git clone https://github.com/pantadeusz/raspigcd
cd raspigcd/
mkdir build
cd build/
cmake ..
make
```

Now I know I should remove older unsupported version - #37

pantadeusz, 2019-10-29:
The v0.9.1.final.archive is the older version. I had my dependency management backwards (I was experimenting with node and the project was not that popular yet so I didn't care much about stable builds). It was on Hackaday, so I decided to keep the older version in subdirectory. Your issue convinced me that It should be left in commit history only, because it makes some confusion.

By the way - I am very curious if you will be able to run the project on Pi Zero. I don't have it and never tested on it. If it would work, I would be verry happy.

spaxxomatic, 2019-10-29:
OK ! I was going for the older version because I also had some compilation problems with the latest. But that was before updating to gcc9, so I'll retry on that one.

spaxxomatic, 2019-10-29:
Oh! v0.9.1  built fine now that the deps are also compiled with g++9 !

spaxxomatic, 2019-10-29:
There are errors building the main version:
```
/root/raspigcd/src/raspigcd.cpp:420:32: error: use of deleted function ‘std::atomic<bool>::atomic(const std::atomic<bool>&)’
  420 |     std::atomic<bool> paused = false;
      |                                ^~~~~
```

pantadeusz, 2019-10-29:
(I was hoping to go to sleep - my time is midnight right now ;) )

This is probably a bug on my side - It works on g++-8 and I didn't test it recently on 9 version. Created #38

Quick fix should be as in https://stackoverflow.com/questions/27314485/use-of-deleted-function-error-with-stdatomic-int

```
// std::atomic<bool> paused = false; // change <-- this to:
std::atomic<bool> paused{false};                      /// <-- this
```

spaxxomatic, 2019-10-29:
For me it's midnight too :) 
I've fixed that also in the meantime - the compile went fine, but I'm hitting now some assembler errors:
Assembler messages:
Error: unknown architecture `armv6kz'

```
Error: unrecognized option -march=armv6kz
CMakeFiles/gcd.dir/build.make:62: recipe for target 'CMakeFiles/gcd.dir/src/raspigcd.cpp.o' failed
make[3]: *** [CMakeFiles/gcd.dir/src/raspigcd.cpp.o] Error 1
CMakeFiles/Makefile2:419: recipe for target 'CMakeFiles/gcd.dir/all' failed
make[2]: *** [CMakeFiles/gcd.dir/all] Error 2

```

spaxxomatic, 2019-10-29:
Can't push the fix - will ask for perms

pantadeusz, 2019-10-29:
You can create pull request on fork, and I will accept it.

The ```-march=armv6kz``` is probably specific to Pi0. Maybe try to remove ```-march=native``` flag from CmakeLists.txt (or other flags).

If it compiles and the machine does not move, then try to comment/uncomment lines https://github.com/pantadeusz/raspigcd/blob/master/src/hardware/driver/raspberry_pi.cpp#L41 and https://github.com/pantadeusz/raspigcd/blob/master/src/hardware/driver/raspberry_pi.cpp#L42

spaxxomatic, 2019-10-30:
Yes, that was it. Set -march=armv6 in CMakeFiles/gcd.dir/flags.make and the build went through
Good night

spaxxomatic, 2019-10-30:
Works  on raspi zero!
Had to enable the #define BCM2708_PERI_BASE       0x20000000   in raspberry_pi.cpp and that was it.

I'd had one question regarding motor config: I have only two steppers and no spindle. If I keep only two motors in the config, will they be the X and Y? Means, only X, Y G-Codes are interpreted? 
Thx

spaxxomatic, 2019-10-30:
BTW: some doc for the API would be very nice.

spaxxomatic, 2019-10-30:
Which OS do you use? Do you have an rt_preempt kernel? 
The jitter with my stock Ubuntu Jessie is pretty bad - the motors are not running smooth.  I have a Xenomai, I should give it a try  ...

pantadeusz, 2019-10-30:
Great to hear that you managed to run this on Pi Zero.

API is on long todo list - I am the only one maintainer of this project and noone is paying me for it, so I have limited resources to write some things (like good documentation).

You can try to increase ```"tick_duration_us": 50``` to something like ```"tick_duration_us": 150```. If you can wait, I can make option to disable multithreaded generation of execution plan parts (it should be possible to reduce fifo size to 1 in https://github.com/pantadeusz/raspigcd/blob/master/src/raspigcd.cpp#L280  -- ```max_queue_size```). Multithreadeading is good when you have multiple cores, but Pi0 have only one core.

I use standard raspbian from Raspberry Pi foundation site.

You can execute gcode that have only X and Y moves, but you should set some pin (unused) for Z (but not use in G-Code).

The design assumptions were that raspigcd would be run on raspberry pi 2 or later with multicore cpu.

BTW: made the modifications that I promised earlier - #38 and #37

spaxxomatic, 2019-10-30:
I could help with the docs and probably also with implementation of new features. 
I increased the tick_duration, that helped, but still not as smooth as PyCNC or other DMA-based methods (https://github.com/Nikolay-Kha/PyCNC). 
PyCNC works pretty nice, but the G-code parsing  and step generation is way too slow, because it's written in python. 
What I intend to do is to use a companion AVR for step generation, (like for exaple Klipper does).
It'll be fairly simple to do it - Raspi would just compute the number of steps and send commands to AVR like "x100steps with freq 1000", "y20 steps...". 
That would allow me to use linear scales for position feedback and have very precise movements.  

BTW, I have a strange problem - the Y axis will not change direction. X is fine. I know everything is wired fine, because PyCNC works with the same config, but gcd will not switch the pin 16 for some reason. Config should be OK: 
` "steppers": [
    {
      "dir": 19,
      "en": 10,
      "step": 16,
      "steps_per_mm": 533.33
    },
    {
      "dir": 20,
      "en": 10,
      "step": 21,
      "steps_per_mm": 533.33
    }
  ],`

Another strange effect  on the X axis - one direction is jerky, the other one is smoother. Do you turn the direction pin on and off at each step (or more often than at the beginning of the move)?

pantadeusz, 2019-10-31:
You should configure third axis to some unused pins. In order to reduce execution branches in the code generating signals to pins I assume there are always 3 motors. You can select unused pins for third motor. The code that drives motors is in https://github.com/pantadeusz/raspigcd/blob/master/src/hardware/driver/raspberry_pi.cpp#L278 - You can see that it always takes 3 coordinates (stepper commands). This is also something that should be in documentation or as a compilation option. There should be also no conflicts between spindle pins and motors (I am quite sure I added exception in the case of conflict).

About DMA - I didn't have enough time to create it. This should be quite easy to achieve in the current project by creating new "driver" similar to the ones in https://github.com/pantadeusz/raspigcd/tree/master/src/hardware/driver .

About contribution. I think that it is a good Idea. On what terms? My default proposition is in readme  but we can discuss other options. I didn't have any requests for commercial license yet (custom license with some additional features in app) so this is probably not going to be very profitable project.

I suggest in the readme that  author's economic rights goes to me, so I would be able to maintain different commercial licenses easily, just in case. I intend to contact contributors in case of some reasonable commercial offer in order to create team and have some profit (and by not doing everything alone, but in team).

pantadeusz, 2019-11-16:
Please let me know if you managed to achieve success? Are there other problems on the way? I would like to close this issue or address other problems.

pantadeusz, 2019-11-21:
The notes from this issue were very good so this topic helped to develop much better code. If no more comments on this, I will close this issue soon.

pantadeusz, 2019-11-28:
This thread seems to be finished. Thank you for good notes. Closing for now.

closed 2019-11-28 by pantadeusz