pantadeusz/raspigcd #62: Question Regarding Development

toozinger, 2020-01-15:
Hello @pantadeusz, 
I am a graduate student at Purdue University, currently looking at control systems for a new custom SLS machine. I have looked into GRBL, TinyG, g2core, LinuxCNC, PyCNC, as well as your raspigcd. I was wondering how the project was going? Is it possible/easy to add custom I/O (such as heater with PID, or on/off controls) or extra axis (not synchronized with x/y/z, but a separate motor that needs to be run sometimes)? Also, how is the stability with high step rates? Thanks! 

Sorry to ask this here instead of a more appropriate location.

pantadeusz, 2020-01-15:
Hi

It is going slowly but steadily. It is not my main field of work, so its development is slower than commercial projects.

Did you Try to run It? Did you have some issues? I am very curious about some feedback (esp stars are coming, but I don't know if anyone actually uses it :) ).

It should be possible to add custom IO.This is just C++ program. Probably it can be put somewhere near https://github.com/pantadeusz/raspigcd/blob/master/src/raspigcd.cpp#L398 . Unfortunately the ```main``` source file needs some refactoring, and I didn't have time to do it properly right now. It should be also possible to increase axis count - the hardware by default have max of 4 steppers - https://github.com/pantadeusz/raspigcd/blob/master/include/hardware/stepping_commands.hpp, the https://github.com/pantadeusz/raspigcd/blob/master/src/hardware/driver/raspberry_pi.cpp#L272 is the place where the steps are executed - it use max 4 motors.

About the stability - it seems very good despite steps execution in regular thread. I did some experiments earlier and I don't expect it to be much worse than that: 

[experiment-250microsecond-signal-measured-by-raspi.png](https://github.com/pantadeusz/raspigcd/blob/v0.9.1.final/doc/experiment-250microsecond-signal-measured-by-raspi.png)

I don't have quick access to oscilloscope right now, so the hardware tests must wait.

toozinger, 2020-01-20:
Hi,

I have not tried it yet but will be ordering a laser engraver soon that I may upgrade to a different control scheme, possibly leveraging this codebase. This project feels more approachable than some of the other larger ones.

Great! Thanks for the good information and links to relevant locations in the code. I'll have to see once I get my machine if I can use the packaged program, or end up overhauling with your code or a GRBL type system.

pantadeusz, 2020-02-04:
Hi

What is the progress in testing. Did you have some issues that should be adressed (except unimplemented curves - this is something on the todo list)

pantadeusz, 2020-03-08:
No updates on the issue. Closing.

closed 2020-03-08 by pantadeusz