pantadeusz/raspigcd #65: The raspigcd.cpp is ugly

pantadeusz, 2020-03-08:
The mentioned file is too big. It should be divided into logical parts.

pantadeusz, 2020-09-03:
Working on it in the https://github.com/pantadeusz/raspigcd/tree/experimental_stepping_rework
I think that this alternative approach will allow for better control over the execution of steps, and it will also increase the processing time for long and slow paths

alizadehreza06, 2022-04-05:
could you send gcd installation file command.
i made git clone https:.....  and others based on your description
it could not find gcd.

pi@raspberrypi:~/raspigcd $ ls
build                   example_config.json  pkg-config.pc.cmake  v1.json
Catch2                  home.gcd             README.md            v2.json
CMakeLists.txt          include              src                  v3.json
configure               installcatch2.sh     tests                v4.json
connection-v3-json.png  LICENSE              t.gcd                v5.json
doc                     off.gcd              thirdparty           v6.json
pi@raspberrypi:~/raspigcd $ gcd -c config_file.json -f program.gcd
bash: gcd: command not found
pi@raspberrypi:~/raspigcd $ gcd -c config_file.json -f home.gcd
bash: gcd: command not found
pi@raspberrypi:~/raspigcd $

pantadeusz, 2022-05-10:
It is in the build directory.

```bash
./build/gcd
```

pantadeusz, 2022-10-20:
Ok. Seems that I can close this issue

closed 2022-10-20 by pantadeusz