pantadeusz/raspigcd #67: Support 5-axis

tlhonmey, 2021-08-29:
There are several projects for open source and/or open hardware 3-axis control.

Unfortunately, when you get to the 5-axis realm, all that seems to be available are proprietary driver boards with proprietary firmware, which only talk to proprietary software, all of which that I've looked at so far only runs on a proprietary OS.

Most of the other CnC control projects are running on microcontroller boards which don't have the capacity to run a 5-axis machine.

So you definitely have an opportunity to differentiate your project here.  

I have a few years of experience at the OS level of stuff like this; so, if you can expand to support 5-axis machines, I can build a customized pi image for it optimized for error-free operation.

pantadeusz, 2021-08-31:
That would be great. Right now I am buried under some other projects and don't have time for developing this one. If you would like to help, especially with 5 axis version, then that would be great. The todo list is very long, the 5 axis is one of the points :)

tlhonmey, 2021-09-06:
Not sure if my math skills are up to the 5axis part, but I can take a look.  Planning to get a newer pi here pretty soon and I'll see what I can do for automatic scheduler and affinity tweaking.

pantadeusz, 2022-10-20:
Stale issue. If something new happens I will reopen.