pantadeusz/raspigcd #68: Support for 3D printer

esuig, 2021-11-16:
Hello, very nice work. I would like to use it to control a 3D printer, is it possible? I think I should add the support for a 4th axis (extruder) and then the controls for the fans, heated bed and hot end (so I should be able to read some input from termocouples and switch on or off the heated bed and hot end). Is it possible somehow? I can add more details if needed

Dronki, 2022-06-17:
I've not tested the project yet, but I would personally start with adding the extruder-stepper on https://github.com/pantadeusz/raspigcd/blob/ce76cc43eb2560c60463f21f4472f34359efd385/src/configuration.cpp#L81
Like `stepper( step-pin, enable-pin, direction-pin, step-amount (?) )`

And after that, identify where the gcode-parsing happens and catch the codes for the extruder.

For the heater, I would implement it using a mosfet-driver that is controlled by a PWM-wave.
Same for the heater-bed.

Dronki, 2022-06-17:
@esuig You could check out this project: [pyCNC](https://pypi.org/project/pycnc/)

pantadeusz, 2022-10-20:
Stale issue. Closing.

closed 2022-10-20 by pantadeusz